githubDante/deye-controller #1: SellProgrammer modbus mapping

VertigoOne1, 2023-05-22:
Evening

I'm playing around with the sellprogrammer as i would like to change grid charge status depending on load shedding stage.

Out of the box, the connecting is working, but i'm getting some interesting return values.

deye-controller$ ./get_mode.py 
____________________________________________________
| Grid  |  Gen  |      Time     |   Pwr    |  SOC % |
|       |       | 00:59 | 22:14 |     2202 |     0% |
|       |       **| 22:14 |** 09:57 |        0 |     0% |
|       |       | 09:57 | 00:00 |     2202 |     0% |
|       |       | 00:00 | 09:57 |     2226 |     0% |
|       |       | 09:57 | 22:26 |        0 |     0% |
|       |       | 22:26 | 00:59 |        0 |     0% |
----------------------------------------------------

and a few seconds later

deye-controller$ ./get_mode.py 
____________________________________________________
| Grid  |  Gen  |      Time     |   Pwr    |  SOC % |
|       |       | 00:59 | 22:30 |     2202 |     0% |
|       |       **| 22:30 | 0**9:57 |        0 |     0% |
|       |       | 09:57 | 00:00 |     2202 |     0% |
|       |       | 00:00 | 09:57 |     2226 |     0% |
|       |       | 09:57 | 22:26 |        0 |     0% |
|       |       | 22:26 | 00:59 |        0 |     0% |
----------------------------------------------------

Everything above is actually invalid.

Obviously your mapping to an inverter you had the ability to test on, so i'm not going to twist your arm, i'm happy to do it myself, but i would like to know what documentation you used to discover the correct registers?! My python skill is not too bad, but i don't actually know any resources that indicate the correct indexes! Would the best way be to just dump everything, and start changing a setting on the inverter and than basically register by register reverse engineer the structure, or is there documentation somewhere as guidance?

I'm on an 8kw sunsynk.

Pleasant evening further!

githubDante, 2023-05-22:
The documentation is named Modbus RTU Protocol (with additional strings in chinese)
I've found the initial version [here](https://github.com/kellerza/sunsynk/issues/63)

I also requested a more recent copy of it directly from DEYE and they provide it. 
It has few additions, it has colored cells, but in general it is the same.
Probably [this repo](https://github.com/kellerza/sunsynk/) is more suitable for the smaller inverters. The registers are
described [here](https://github.com/kellerza/sunsynk/blob/main/sunsynk/definitions.py)

I don't know which documentation they used (it's probably listed somewhere in the repo).

As you can see there, the programing registers are in the range 250-279. Here they are between 148-177 and
the days of the week for which the program must be applied is controlled by register 146.

VertigoOne1, 2023-05-24:
Thank you very much! I'll start digging into it.

closed 2023-05-24 by VertigoOne1