githubDante/deye-controller #2: struct.error: 'h' format requires -32768 <= number <= 32767 -> value is 32851

bobo71dream, 2023-07-13:
Hello,

since yesterday i get this error. Nothing was changed by me:

reading of this value get the follow error: TotalSellGrid = LongType(524, 'total_sold_to_grid', 10, suffix='kWh')

DEBUG:pysolarmanv5.pysolarmanv5:Socket setup completed... <socket.socket fd=3, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('10.10.0.48', 59274), raddr=('10.10.0.49', 8899)>
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 55 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 71 00 01 d5 a9 38 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 55 10 17 7c 4a a2 02 01 c2 c1 56 00 6e 1a 00 00 c2 c5 58 64 01 03 02 fd 82 78 b5 77 15
DEBUG:pysolarmanv5.pysolarmanv5:Socket setup completed... <socket.socket fd=3, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('10.10.0.48', 59282), raddr=('10.10.0.49', 8899)>
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 54 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 01 84 0a d4 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 54 11 17 7c 4a a2 02 01 c3 c1 56 00 6f 1a 00 00 c2 c5 58 64 01 03 02 00 05 78 47 91 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 55 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 01 d5 ca e7 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 55 12 17 7c 4a a2 02 01 c3 c1 56 00 6f 1a 00 00 c2 c5 58 64 01 03 02 00 01 79 84 cd 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 56 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 05 75 c9 8d 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 1d 00 10 15 56 13 17 7c 4a a2 02 01 c3 c1 56 00 6f 1a 00 00 c2 c5 58 64 01 03 0a 32 32 31 32 30 32 38 30 31 39 fd 20 f9 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 57 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 08 00 01 05 c8 1e 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 57 14 17 7c 4a a2 02 01 c3 c1 56 00 6f 1a 00 00 c2 c5 58 64 01 03 02 00 00 b8 44 cf 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 58 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 3e 00 03 64 07 f5 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 19 00 10 15 58 15 17 7c 4a a2 02 01 c3 c1 56 00 6f 1a 00 00 c2 c5 58 64 01 03 06 17 07 0d 07 0b 18 23 c5 1a 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 59 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 4a 00 01 a5 dc 16 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 59 16 17 7c 4a a2 02 01 c3 c1 56 00 6f 1a 00 00 c2 c5 58 64 01 03 02 00 01 79 84 d5 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 5a 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 50 00 01 84 1b 3b 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 5a 17 17 7c 4a a2 02 01 c3 c1 56 00 6f 1a 00 00 c2 c5 58 64 01 03 02 00 01 79 84 d7 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 5b 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 62 00 01 25 d4 a8 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 5b 18 17 7c 4a a2 02 01 c3 c1 56 00 70 1a 00 00 c2 c5 58 64 01 03 02 00 00 b8 44 d8 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 5c 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 66 00 01 64 15 2d 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 5c 19 17 7c 4a a2 02 01 c3 c1 56 00 70 1a 00 00 c2 c5 58 64 01 03 02 01 18 b9 de 8e 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 5d 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 68 00 01 05 d6 92 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 5d 1a 17 7c 4a a2 02 01 c3 c1 56 00 70 1a 00 00 c2 c5 58 64 01 03 02 00 19 79 8e 00 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 5e 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 6c 00 01 44 17 17 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 5e 1b 17 7c 4a a2 02 01 c3 c1 56 00 70 1a 00 00 c2 c5 58 64 01 03 02 00 69 78 6a 2d 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 5f 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 6d 00 01 15 d7 aa 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 5f 1c 17 7c 4a a2 02 01 c3 c1 56 00 70 1a 00 00 c2 c5 58 64 01 03 02 00 87 f8 26 89 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 60 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 6f 00 01 b4 17 8c 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 60 1d 17 7c 4a a2 02 01 c4 c1 56 00 70 1a 00 00 c2 c5 58 64 01 03 02 00 00 b8 44 e3 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 61 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 72 00 01 24 11 fa 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 61 1e 17 7c 4a a2 02 01 c4 c1 56 00 70 1a 00 00 c2 c5 58 64 01 03 02 03 de 38 ec ee 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 62 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 73 00 01 75 d1 0d 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 62 1f 17 7c 4a a2 02 01 c4 c1 56 00 70 1a 00 00 c2 c5 58 64 01 03 02 00 0a 38 43 70 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 63 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 74 00 01 c4 10 9d 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 63 20 17 7c 4a a2 02 01 c4 c1 56 00 70 1a 00 00 c2 c5 58 64 01 03 02 00 19 79 8e 0d 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 64 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 75 00 01 95 d0 30 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 64 21 17 7c 4a a2 02 01 c4 c1 56 00 70 1a 00 00 c2 c5 58 64 01 03 02 00 14 b8 4b 06 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 65 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 76 00 01 65 d0 02 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 65 22 17 7c 4a a2 02 01 c4 c1 56 00 70 1a 00 00 c2 c5 58 64 01 03 02 12 c0 b4 b4 2b 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 66 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 77 00 01 34 10 13 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 66 23 17 7c 4a a2 02 01 c4 c1 56 00 71 1a 00 00 c2 c5 58 64 01 03 02 14 00 b7 44 03 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 67 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 78 00 01 04 13 e8 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 67 24 17 7c 4a a2 02 01 c4 c1 56 00 71 1a 00 00 c2 c5 58 64 01 03 02 13 60 b5 5c 7a 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 68 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 7e 00 01 e4 12 ce 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 68 25 17 7c 4a a2 02 01 c4 c1 56 00 71 1a 00 00 c2 c5 58 64 01 03 02 13 24 b5 6f 53 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 69 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 7f 00 01 b5 d2 61 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 69 26 17 7c 4a a2 02 01 c4 c1 56 00 71 1a 00 00 c2 c5 58 64 01 03 02 00 1e 38 4c 9c 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 6a 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 80 00 01 85 e2 43 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 6a 27 17 7c 4a a2 02 01 c4 c1 56 00 71 1a 00 00 c2 c5 58 64 01 03 02 00 28 b8 5a 36 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 6b 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 8f 00 01 b5 e1 82 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 6b 28 17 7c 4a a2 02 01 c5 c1 56 00 71 1a 00 00 c2 c5 58 64 01 03 02 01 f4 b8 53 ff 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 6c 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 bd 00 01 14 2e 5d 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 6c 29 17 7c 4a a2 02 01 c5 c1 56 00 71 1a 00 00 c2 c5 58 64 01 03 02 00 01 79 84 ff 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 6d 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 c0 00 01 84 36 d9 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 6d 2a 17 7c 4a a2 02 01 c5 c1 56 00 71 1a 00 00 c2 c5 58 64 01 03 02 00 64 b9 af cf 15
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 6e 00 17 7c 4a a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 c1 00 01 d5 f6 ec 15
[... cut ...]
    data_dict[iteratiorentry.description]=convertinttofloat(iteratiorentry.format())
  File "/usr/script/deye_controller/modbus/protocol.py", line 86, in format
    v = to_bytes(self.value[::-1])
  File "/usr/script/deye_controller/modbus/utils.py", line 16, in to_bytes
    return bytes.fromhex(''.join([struct.pack('>h', x).hex() for x in val]))
  File "/usr/script/deye_controller/modbus/utils.py", line 16, in <listcomp>
    return bytes.fromhex(''.join([struct.pack('>h', x).hex() for x in val]))
struct.error: 'h' format requires -32768 <= number <= 32767`

githubDante, 2023-07-13:
Definitely a bug. It will need a new type LongUnsigned. The actual value  is 3285.1

bobo71dream, 2023-07-13:
i have make a local try except to skip the error. How can we fix this?

githubDante, 2023-07-13:
The value is definitely a float as it should be multiplied by 0.1 / divided by 10 in order to be converted to kWh.

It should be OK now . Thanks for reporting it.

bobo71dream, 2023-07-13:
I have tested, woks perfect. Thx for your fast support.

closed 2023-07-13 by githubDante