githubDante/deye-controller #4: Cannot read register 145 Solar Sell

tpahula, 2023-08-24:
I am trying to read and write register number 145. I will get following error:

 deye-regtest 192.168.1.x 2705xxxx 145

raise IllegalDataValueError('Quantify field of request must be a '
umodbus.exceptions.IllegalDataValueError:  The value contained in the request data field is not an allowable value
    for the server.

githubDante, 2023-08-24:
What's the second value used in the command 146 ?!? I must admit that the `--help` of this tool is not correct `end` should be number of registers to be read. Try with 
```
deye-regtest 192.168.1.x 2705xxxx 145 1
```

and you should get the value (0 is the default I think)

tpahula, 2023-08-24:
Ok. It is working now. I can change Solar Sell value with commands:

Set Solar Sell enabled:

deye-regwrite 192.168.1.x 2705xxxxxx 145 1

Set Solar Sell disabled:

deye-regwrite 192.168.1.x 2705xxxxxx 145 0

Modbus RTU documentation is available from:

https://community.openhab.org/uploads/short-url/vxKms3k37I8n0AoaQIT2mW25eNb.pdf

How I can use deye-controller with Home Assistant ? I am using deye-controller in my OpenWRT router now.

githubDante, 2023-08-28:
You can't, unless a Home Assistant plugin is developed around it. You can use it for a convenient reading from / writing to the inverter. 
My personal use case is data collection and upload to a PostgreSQL DB + remote control of the inverter(s).

closed 2023-08-30 by githubDante