githubDante/deye-controller #5: Documentation: more info on usage of the code

chinezbrun, 2023-08-31:
is a good approach and i will like to use it in my automation but i need more info on the usage of the code.

i.e in example provided the is a syntax like:
>>> register = HoldingRegisters.BMSBatteryCapacity
BMSBatteryCapacity is an example but where we can find the rest?

similar for WritableRegisters :)

githubDante, 2023-08-31:
All registers are accessible as class properties and are listed even in the REPL on  Tab+Tab

```

>>> from deye_controller.modbus.protocol import HoldingRegisters, WritableRegisters
>>> HoldingRegisters.BMS<Tab><Tab>
HoldingRegisters.BMSBatteryAlarm           HoldingRegisters.BMSBatterySOH             HoldingRegisters.BMSChargedVoltage         HoldingRegisters.BMSMaxChargingCurrent
HoldingRegisters.BMSBatteryCapacity        HoldingRegisters.BMSBatterySymbol          HoldingRegisters.BMSChargingCurrentLimit   HoldingRegisters.BMSMaxDischargeCurrent
HoldingRegisters.BMSBatteryCurrent         HoldingRegisters.BMSBatteryTemp            HoldingRegisters.BMSDischargeCurrentLimit  HoldingRegisters.BMSType
HoldingRegisters.BMSBatteryFaultLocation   HoldingRegisters.BMSBatteryVoltage         HoldingRegisters.BMSDischargedVoltage      

```

On read the values received from the inverter must set as a value to the respective register e.g.

`>>> HoldingRegisters.BMSBatteryTemp.value = 1410 
`

You can use the *group_registers()* / *map_response()* helpers from _utils_ to do that. The actual (human readable) values are returned from a call to the format() method e.g.
```

>>> HoldingRegisters.BMSBatteryTemp.format()
41.0

```
For writing the values are set via the set() method e.g.

```
>>> WritableRegisters.SellModeVolts1.set(51)
```

and then the actual modbus values which must be sent to the inverter are exposed as a property

```
>>> WritableRegisters.SellModeVolts1.to_modbus
5100
```

chinezbrun, 2023-09-01:
thanks for your answer. 
indeed if you ar digging you can find them.  the point was how to find info faster. but anyhow i get the point,
i understand that you are upload the parameters on PostgreSQL. How?  I have MariaDB and 
and i'm looking to have them there.

githubDante, 2023-09-01:
How ... A call to a DB stored procedure (via asyncpg) with the collected data. The data is stored in the following format:

`<date> | <inverter serial: numeric> | <metric_1: double precision[]> | <metric_2: double precision[]> | ...`

Each index in the metric arrays corresponds to an hour/minute integer and is defined by:

```
    current_t := now();
    h_idx := extract(hour from current_t);
    m_idx := extract(minute from current_t);
    
    _idx_ := h_idx * 12 + m_idx / 5;
```
    
where the variables are declared as:
    
```
    declare h_idx int;
    declare m_idx int;
    declare _idx_ int;
    declare current_t timestamp without time zone;
```

This gives you 288 records/24h for each metric (column) and everything is stored in a single row, so you have really quick access to the data for the respective day.
MariaDB doesn't seem like the right tool for the task as it does not support arrays as far as I know.

chinezbrun, 2023-09-02:
thanks! do your script could provide output data in JSON format?

githubDante, 2023-09-05:
Not currently, but could be added easily (assuming we are talking about `deye-read`). Any suggestions about the JSON structure ?

githubDante, 2023-09-15:
JSON output is supported in version 0.1.3. The output is available in the [examples dir ](https://github.com/githubDante/deye-controller/blob/master/examples/all.json)

chinezbrun, 2023-09-15:
wow. super fast. looks good. 
sorry that i did not have the chance to provide you a suggestion for JSON structure...
but instead of: 
```
 {
  "logger": 2719999999,
  "serial": "2299999999",
  "data": [
  {
      "device_type": {
        "addr": 0,
        "value": "Hybrid3Phase",
        "unit": ""
      }
  }
  ,
    {
      "modbus_address": {
        "addr": 1,
        "value": 1,
        "unit": ""
      }
    }
  
  ]
}
```
why not:
```
{
  "logger": 2719999999,
  "serial": "2299999999",
  "data": [{
      "device_type": {
        "addr": 0,
        "value": "Hybrid3Phase",
        "unit": ""
      }
  ,
      "modbus_address": {
        "addr": 1,
        "value": 1,
        "unit": ""
      }
  }
]
}
```
and i believe will simplify the future reading of JSON data

githubDante, 2023-09-15:
Current implementation allows you to read the output like this:
```
for item in output['data']:
    for k,v in item.items():
        if v['addr'] in [145, 214]:
            do_something_with(v['value'])
```
or if filtering by description is required:
```

for item in data['data']:
    for k,v in item.items():
        if k in ['solar_sell', 'bms_battery_SOC']:
            do_something_with(v['value'])
     
``` 
I don't see a problem with that. A loop over the ['data'] key would be needed anyway.

chinezbrun, 2023-09-15:
clear. where is the file path? can be changed?

githubDante, 2023-09-15:
Currently the output is written directly to `stdout`. On Linux it's easy to write it wherever you want by using `> <path>/<filename>` but probably we are talking for universal method for all platforms.

Expect an `--out` option in 0.1.4

chinezbrun, 2023-09-15:
cool.good job!

githubDante, 2023-09-15:
Do you know how to update from git in order to test this [d91f755](https://github.com/githubDante/deye-controller/commit/d91f75571434d249a56d1e5bd71b02fc3f0ce981)  :)

chinezbrun, 2023-09-16:
done. what's the sintax?

githubDante, 2023-09-18:
Sorry. Apparently update directly from git does not update the console scripts. Version 0.1.4 is already on pypi.org

`pip install --upgrade deye-controller` 

and should  be OK. The syntax is 

`deye-read --json --out <filename> <ip_addr> <logger_sn> `

chinezbrun, 2023-09-20:
> Sorry. Apparently update directly from git does not update the console scripts. Version 0.1.4 is already on pypi.org
> 
> `pip install --upgrade deye-controller`
> 
> and should be OK. The syntax is
> 
> `deye-read --json --out <filename> <ip_addr> <logger_sn> `

it is working as expected!

chinezbrun, 2023-10-14:
hello.
i can't find the Gen Port data in the JSON file . 
meaning: voltage, power per phases and total power. registries 661 - 667.
this is useful when you have the microinverter coupled.

githubDante, 2023-10-15:
Yes, all generator related registers are missing (saves time when there so many others). I'll try to add them in the next few days.

githubDante, 2023-10-16:
Generator registers have been added. The documentation is little fishy about these. Let me know if there are any problems with them.

chinezbrun, 2023-10-17:
ok. i need some good sunny days to test. do feel the need to open a separate issue on this topic?

chinezbrun, 2024-01-11:
I confirm that microinverter output (generator) is displayed correct.
[image]