githubDante/deye-controller #7: Possibility and example for change Gen Port Workin Mode

bobo71dream, 2023-11-05:
Hello,

ich search for a solution to change with a time based script the Gen Port Morking Mode from "MicrInv Input" to "Genrator Input". It is possible to write this with your controller?
The benefit is to reduse the grid usage by night to zero.

Thank you.

githubDante, 2023-11-06:
If we talk about this screen

[image]

probably it is controlled by register 133. The accepted values should be 0, 1 and 2: 

 0 - for generator input
 1 - for smart load
 2 - for micro inverter

Currently the register is not mapped here. However, it can be tested by changing the selection manually and checking the value of the register with:
`
 deye-regtest <ip addr> <logger serial> 133 1
 `
Can you confirm that the values above are correct.

bobo71dream, 2023-11-07:
Thank you for your example. But where i can find deye-regtest? Is not in your code...

githubDante, 2023-11-07:
It's available as a cli command after [install](https://github.com/githubDante/deye-controller#installl)

bobo71dream, 2023-11-07:
Okay, i have found the comand. But by running i get this error on 2 different pi´s.
```
`root@nas-teslamate-teslasolarcharger:~# deye-regtest '10.10.0.49', 2722790423 133 1
Traceback (most recent call last):
  File "/usr/local/bin/deye-regtest", line 8, in <module>
    sys.exit(test_register())
  File "/usr/local/lib/python3.9/dist-packages/deye_controller/deye_reader.py", line 117, in test_register
    _read_registers(opt.address, opt.serial, opt.start, opt.count)
  File "/usr/local/lib/python3.9/dist-packages/deye_controller/deye_reader.py", line 68, in _read_registers
    inv = PySolarmanV5(address, int(logger_serial), port=8899, mb_slave_id=1, verbose=False, socket_timeout=10)
  File "/usr/local/lib/python3.9/dist-packages/pysolarmanv5/pysolarmanv5.py", line 118, in __init__
    self._socket_setup(kwargs.get("socket"), kwargs.get("auto_reconnect", False))
  File "/usr/local/lib/python3.9/dist-packages/pysolarmanv5/pysolarmanv5.py", line 404, in _socket_setup
    raise NoSocketAvailableError("No socket available")
pysolarmanv5.pysolarmanv5.NoSocketAvailableError: No socket available
root@nas-teslamate-teslasolarcharger:~#
```
How can i fix it?

githubDante, 2023-11-07:
Don't use quotes and commas. The command should be:
```
deye-regtest 10.10.0.49 2722790423 133 1
```

bobo71dream, 2023-11-07:
> deye-regtest 10.10.0.49 2722790423 133 1

Oh shit, that was the solution. Many many thank you. The result is [2] ->  micro inverter
```
root@nas-teslamate-teslasolarcharger:~# deye-regwrite 10.10.0.49 2722790423 133 0
Wrote: 1
root@nas-teslamate-teslasolarcharger:~# deye-regtest 10.10.0.49 2722790423 133 1
[0]
root@nas-teslamate-teslasolarcharger:~#
```
Writing from console also work. Many thx. But last question. How can i write the register in python? I have found the functions. But for me is not clear, how can i get the parameter to the function. Can you explain me this?

githubDante, 2023-11-07:
Thank you for the confirmation.

As for the mapping, you need Google Lens and the inverter MODBUS RTU documentation.

githubDante, 2024-03-11:
The register is now  exposed for reading & writing in a more programmable way. See d39aa6f  & c84c23c

closed 2024-03-11 by githubDante