githubDante/deye-controller #9: Many issues to a mega tool

europer, 2023-12-23:
First of all this looks fantastic even though I come with a lot of questions (sorry). I can now see that this will work. First, a prologue where I am standing and my approach.

- I have a 12 kWp Sungrow plant with a Sungrow 16 kW battery for storage.

- I use Home Assistant and Node-Red to control my house and the Sungrow PV plant. At a given time, I can force my Sungrow over a ModBus protocol to discharge/charge since I use Tibber and buy cheap power from October to March when needed at the right time. The rest of the year, the system works on self-consumption since I have a surplus anyway, and the inverter puts the rest into the battery (work still needs to be done to make the coding in Node-Red a bit nicer since I'm not a programmer but a DIY man).

- Now, I have the Deye Sun 12k with 4x5.12 kWh (44-57V) installed in the garage, now for the 24/7/365 base load. The Sungrow will take the peaks, heavy loads, and the wallbox since it is just faster and can deliver up to 11 kW directly.

- I use the integration from Solman in Home Assistant. So, for the Deye, I would like to take control like the Sungrow and control when to charge/discharge. I tried with ESP32 and Modbus (https://github.com/klatremis/esphome-for-deye) but tried everything I know (changed the tx/rx, 1000 times the Modbus pins), then with Elfin ew11 Modbus2Wifi and tried the https://kellerza.github.io/sunsynk/reference/multi-options, even with a different Elfin Modbus2Wifi but the same setup, no connection. Same integration from Kellerza. Then it is possible to use a Raspi or SBC, also to communicate USB2Modbus, so I did the setup, but I use Dietpi as a derivative for the Linux/Debian, so even my RasPi 4 or my NanoPi R2S, both with Dietpi Debian, now I get a problem with the installation of cmake. Here's the cooking menu, and it stops at point 6.

1. sudo apt-get install git
2. sudo apt-get install cmake
3. git clone https://github.com/3cky/mbusd.git mbusd.git
4. cd mbusd.git
5. mkdir -p build && cd build
6. cmake -DCMAKE_INSTALL_PREFIX=/usr ..
7. make
8. sudo make install
9. sudo nano /etc/mbusd/mbusd.conf.example
10. systemctl start mbusd@ttyUSB0.service
(nicht zwingend notwendig)11. journalctl -u mbusd@ttyUSB0.service -f -n 10
12. sudo systemctl enable mbusd@ttyUSB0.service

And here it would be possible to control the Deye. But at given time Im stopped here.................

Now, I saw your article here, and in Home Assistant, I use the integration Solarman and the WiFi dongle at my Deye. It works perfectly and can SEE all values but can't send commands to control my Deye.

So, here are my questions (every start is hard).

1. On what system does this Deye control run (PC, Raspi),
2. Is it Linux/Debian?
3. Can I use Node-Red for this? Or how do I communicate with your code so that I can use it for my issues?

Here are some pics where I would love to control the Deye (the rest for monitoring is the HA from Solarman integrated).

So my goal is to change the values over the system work mode 
Work Mode1: Selling First, Zero Export to laod/Ct and Energy pattern (batt/load first)
Work Mode2: all parameters

Battery Settings, Batt Set 2 the Ampere (at me the 80A in the pic) and the Grid charge and Grid Signal.

Thanks in advance and have a marry xmas.

BR
Europer

[image]
[image]
[image]
[image]

githubDante, 2024-01-02:
Hi @europer,

Sorry for the late resonse. Directly on your questions:

 - 1 & 2 - The inverters are controlled through an OrangePi running on Raspbian
 - 3 - Don't know about NodeRed. If Python code can be executed from NodeRed probably scripts with input parameters can be used.
 
 Here what I do to control the inverter(s). I use a daemon to download the settings from a remote server, the daemon then compares the current settings of the invreter with those received from the server and update only what's necessary. Dynamic controls (in my usecase) are needed only for the GridExport and the charge/discharge schedule (System Work Mode in the first snapshot). 
 
 As for the second part of your post: 
   - Work Mode1 (snapshot 2) - the available documentation is very vague about them. The max sell power field is controlled by register 143 I think (not defiend as writable ATM), the other settings on the screen are probably close, but testing is needed in order the actual registers to be confirmed.  
   - Work Mode2 (snapshot 1) - all parameters can be controlled remotely. You can use the [SellProgrammer ](https://github.com/githubDante/deye-controller/blob/360856987a4f57f4280face10ed7c2e24668e381/deye_controller/sell_programmer.py#L9) class for that.

europer, 2024-01-18:
Sorry for late respone my self. I will try it, anyway thanks .

tmbreuker, 2025-05-22:
@europer  I am looking for a similar intergration between HA and my Deye - I stumbled upon this post and also this other github thread: https://github.com/klatremis/esphome-for-deye/issues/47 . Where you able to create a working solution which you could share?