githubDante/deye-controller #10: deye-read

sante85, 2024-02-03:
Hi @githubDante 
I have launched these commands

 deye-read 192.168.0.105 2726848581
get all parameters to 0 value. why?

deye-scan 192.168.0.255

        LoggerSN:   2726848581
        IP:         192.168.0.105
        MAC:        E8FDF879E708

and subsequent

 deye-read 192.168.0.105 2726848581
[                        Device_Type]: Hybrid
[                     Modbus_Address]: 1
[                    Inverter_Serial]: 2212132808
[                        Rated_Power]: 0
[                      Inverter_Time]: 1970-01-01 00:00:00
[                       Comm_Address]: 6685
[                      Switch_On_Off]: Inactive
[                       Control_Mode]: LeadBattery
[                Batt_Equalization_V]: 0.0 V
[                  Batt_Absorbtion_V]: 20.49 V
[                       Batt_Float_V]: 0.0 V
[                      Batt_Capacity]: 0 Ah
[                       Batt_Empty_V]: 0.0 V
[                  Zero_Export_Power]: 0 W
[                             Tempco]: 565 mV/*C
[                    Max_Charge_Amps]: 78 A
[                 Max_Discharge_Amps]: 21 A
[               Battery_Control_Mode]: Error
[                    Battery_Wake_Up]: Inactive
[                 Battery_Resistance]: 0 mOhm
[               Battery_Charging_Eff]: 0.0 %
[          Battery_Shutdown_Capacity]: 0 %
[          Battery_Recovery_Capacity]: 0 %
[               Battery_Low_Capacity]: 0 %
[           Battery_Shutdown_Voltage]: 0.0 V
[            Battery_Restart_Voltage]: 378.89 V
[                Battery_Low_Voltage]: 4.18 V
[          Grid_Charge_Start_Voltage]: 55.0 V
[              Grid_Charge_Start_Soc]: 0 %
[                Grid_Charge_Current]: 0 A
[                Grid_Max_Output_Pwr]: 5394 W
[                         Solar_Sell]: Inactive
[                   Sell_Time_Of_Use]: ['TimeOfUse - ENABLED', 'TimeOfUse - TUESDAY', 'TimeOfUse - THURSDAY']
[                      Sell_Point_T1]: 00:90
[                      Sell_Point_T2]: 24:73
[                      Sell_Point_T3]: 00:52
[                      Sell_Point_T4]: 00:00
[                      Sell_Point_T5]: 00:52
[                      Sell_Point_T6]: 00:00
[                Sell_Point_T1_Watts]: 2401 W
[                Sell_Point_T2_Watts]: 0 W
[                Sell_Point_T3_Watts]: 2401 W
[                Sell_Point_T4_Watts]: 2474 W
[                Sell_Point_T5_Watts]: 0 W
[                Sell_Point_T6_Watts]: 0 W
[                Sell_Point_T1_Volts]: 0.0 V
[                Sell_Point_T2_Volts]: 0.0 V
[                Sell_Point_T3_Volts]: 0.0 V
[                Sell_Point_T4_Volts]: 0.0 V
[                Sell_Point_T5_Volts]: 2.1 V
[                Sell_Point_T6_Volts]: 0.0 V
[                  Sell_Point_T1_Soc]: 0 %
[                  Sell_Point_T2_Soc]: 0 %
[                  Sell_Point_T3_Soc]: 0 %
[                  Sell_Point_T4_Soc]: 0 %
[                  Sell_Point_T5_Soc]: 0 %
[                  Sell_Point_T6_Soc]: 0 %
[                    Charge_Point_T1]: GridGenDisabled
[                    Charge_Point_T2]: GridGenDisabled
[                    Charge_Point_T3]: GridGenDisabled
[                    Charge_Point_T4]: GridGenDisabled
[                    Charge_Point_T5]: GridGenDisabled
[                    Charge_Point_T6]: GridGenDisabled
[              Export_To_Grid_Cutoff]: {'enabled': <TwoBitState.Undefined_1: 0>, 'generator_peak_shaving': <TwoBitState.Undefined_2: 1>, 'grid_peak_shaving': <TwoBitState.Undefined_2: 1>, 'on_grid_always_on': <TwoBitState.Disabled: 2>, 'external_relay': <TwoBitState.Undefined_2: 1>, 'report_loss_of_lithium': <TwoBitState.Undefined_1: 0>}
[                Grid_Freq_Selection]: 60 Hz
[                  Grid_High_Voltage]: 40.4 V
[                   Grid_Low_Voltage]: 0.0 V
[                Grid_High_Frequency]: 0.0 Hz
[                 Grid_Low_Frequency]: 0.0 Hz
[                  Generator_To_Grid]: Inactive
[               Gen_Peak_Shaving_Pwr]: 434 W
[              Grid_Peak_Shaving_Pwr]: 837 W
[              Smart_Load_Open_Delay]: 5000 Minutes
[                Output_Power_Factor]: 500.0 %
[                Bms_Charged_Voltage]: 1.0 V
[             Bms_Discharged_Voltage]: 1.3 V
[           Bms_Charge_Current_Limit]: 0 A
[        Bms_Discharge_Current_Limit]: 0 A
[                    Bms_Battery_Soc]: 1 %
[                Bms_Battery_Voltage]: 0.19 V
[                Bms_Battery_Current]: 990 A
[                   Bms_Battery_Temp]: -98.0 °C
[             Bms_Max_Charge_Current]: 35 A
[          Bms_Max_Discharge_Current]: 35 A
[                  Bms_Battery_Alarm]: Inactive
[         Bms_Battery_Fault_Location]: 5250
[               Bms_Battery_Symbol_2]: 5100
Unknown CAN/MODBUS protocol: 80
[                           Bms_Type]: PYLONTech_CAN
[                    Bms_Battery_Soh]: 0
[                 Max_Solar_Sell_Pwr]: 10 W
[                     Inverter_State]: StandBy
[                 Active_Power_Today]: 0.0 kWh
[               Reactive_Power_Today]: 0.0 kVarh
[              Grid_Connection_Today]: 0.0 minutes
[               Battery_Charge_Today]: 0.0 kWh
[            Battery_Discharge_Today]: 0.0 kWh
[               Battery_Charge_Total]: 0.0 kWh
[            Battery_Discharge_Total]: 0.0 kWh
[             Today_Bought_From_Grid]: 0.0 kWh
[                 Today_Sold_To_Grid]: 0.0 kWh
[             Total_Bought_From_Grid]: 0.0 kWh
[                 Total_Sold_To_Grid]: 0.0 kWh
[                      Today_To_Load]: 0.0 kWh
[                      Total_To_Load]: 0.0 kWh
[                      Today_From_Pv]: 0.0 kWh
[                   Today_From_Pv_S1]: 0.0 kWh
[                   Today_From_Pv_S2]: 0.0 kWh
[                      Total_From_Pv]: 0.0 kWh
[                Battery_Temperature]: 0.0 °C
[                    Battery_Voltage]: 0.0 V
[                        Battery_Soc]: 0.0 %
[                  Battery_Out_Power]: 0 W
[                Battery_Out_Current]: 0.0 A
[               Battery_Corrected_Ah]: 0 Ah
[                  Grid_Phase_A_Volt]: 0.0 V
[                  Grid_Phase_B_Volt]: 0.0 V
[                  Grid_Phase_C_Volt]: 0.0 V
[                 Grid_Phase_Ab_Volt]: 0.0 V
[                 Grid_Phase_Bc_Volt]: 0.0 V
[                 Grid_Phase_Ca_Volt]: 0.0 V
[              Grid_Phase_A_In_Power]: 0 W
[              Grid_Phase_B_In_Power]: 0 W
[              Grid_Phase_C_In_Power]: 0 W
[     Grid_Active_Side_Side_In_Power]: 0 W
[Grid_Active_Side_Side_Apparent_Power]: 0 W
[                  Grid_In_Frequency]: 0.0 Hz
[            Grid_Phase_A_In_Current]: 0.0 A
[            Grid_Phase_B_In_Current]: 0.0 A
[            Grid_Phase_C_In_Current]: 0.0 A
[   Grid_Phase_A_Out_Of_Grid_Current]: 0.0 A
[   Grid_Phase_B_Out_Of_Grid_Current]: 0.0 A
[   Grid_Phase_C_Out_Of_Grid_Current]: 0.0 A
[     Grid_Phase_A_Out_Of_Grid_Power]: 0 W
[     Grid_Phase_B_Out_Of_Grid_Power]: 0 W
[     Grid_Phase_C_Out_Of_Grid_Power]: 0 W
[       Grid_Total_Out_Of_Grid_Power]: 0 W
[Grid_Total_Out_Of_Grid_Apparent_Power]: 0 W
[                Grid_Phase_A__Power]: 0 W
[                Grid_Phase_B__Power]: 0 W
[                Grid_Phase_C__Power]: 0 W
[                   Grid_Total_Power]: 0 W
[              Grid_Phase_A_Volt_Out]: 0.0 V
[              Grid_Phase_B_Volt_Out]: 0.0 V
[              Grid_Phase_C_Volt_Out]: 0.0 V
[       Inverter_Phase_A_Out_Current]: 0.0 A
[       Inverter_Phase_B_Out_Current]: 0.0 A
[       Inverter_Phase_C_Out_Current]: 0.0 A
[         Inverter_Phase_A_Out_Power]: 0 W
[         Inverter_Phase_B_Out_Power]: 0 W
[         Inverter_Phase_C_Out_Power]: 0 W
[           Inverter_Total_Out_Power]: 0 W
[  Inverter_Total_Apparent_Out_Power]: 0 W
[             Inverter_Out_Frequency]: 0.0 Hz
[                  Ups_Phase_A_Power]: 0 W
[                  Ups_Phase_B_Power]: 0 W
[                  Ups_Phase_C_Power]: 0 W
[                    Ups_Total_Power]: 0 W
[                  Load_Phase_A_Volt]: 0.0 V
[                  Load_Phase_B_Volt]: 0.0 V
[                  Load_Phase_C_Volt]: 0.0 V
[               Load_Phase_A_Current]: 0.0 A
[               Load_Phase_B_Current]: 0.0 A
[               Load_Phase_C_Current]: 0.0 A
[                 Load_Phase_A_Power]: 0 W
[                 Load_Phase_B_Power]: 0 W
[                 Load_Phase_C_Power]: 0 W
[                   Load_Total_Power]: 0 W
[                       Pv1_In_Power]: 0 W
[                       Pv2_In_Power]: 0 W
[                           Pv1_Volt]: 0.0 V
[                        Pv1_Current]: 0.0 A
[                           Pv2_Volt]: 0.0 V
[                        Pv2_Current]: 0.0 A

githubDante, 2024-02-04:
Hi @sante85 

Unfortunately, your inverter uses a different protocol/mapping. At least it is recognized properly (`[ Device_Type]: Hybrid`). Do you have access to the MODBUS documentation for this device?!?

sante85, 2024-02-04:
mmh, I don't know if exist. Can you suggest other python project that implement this read?

githubDante, 2024-02-04:
Try with https://github.com/kellerza/sunsynk

closed 2024-03-10 by githubDante

sante85, 2024-03-10:
I want use a python script without wrapper of homeassistant.
Only pure python script

pentex017, 2024-05-02:
I managed to get the equivalent register numbers from sunsynk definition.