githubDante/deye-controller #11: feature request - AC temperature

ilia92, 2024-03-10:
First of all I'd like to say thank you for this software. 

I see the temperature of the AC side in the app, so the inverter must have it as information. 

Can this information be provided in the **deye-read** output ?

[image]

githubDante, 2024-03-10:
Hi,

The registers are already in the master branch and will be listed by `deye-read` in the next version. The AC temp should correspond to the [HeatsinkTemp](https://github.com/githubDante/deye-controller/blob/360856987a4f57f4280face10ed7c2e24668e381/deye_controller/modbus/protocol.py#L489)

githubDante, 2024-03-11:
Version 0.1.5 has been published, the heatsink temp should be visible after upgrade.

closed 2024-03-11 by githubDante

ilia92, 2024-03-12:
@githubDante thank you for your very fast response. Works perfect