githubDante/deye-controller #12: Attempting to make it work with my SUN-6K-SG03LP1-EU

CidiRome, 2024-04-23:
Hello.

I have one SUN-6K-SG03LP1-EU with one Pylontech 5000 and I'm trying to use this code to read and write from/to the inverter.

I'm having a bit of a hard time understanding how it work as I don't recognize the language/coding show in the examples despite having some programming background (mostly C).

I managed to install it, but doesn't seem to be reading the inverter correctly:

I read the inverter serial number correctly, so it is working up to some level.

`[dividi@rocky deye-controller-master]$ deye-read 192.168.192.12 2913021234
[                        Device_Type]: Hybrid
[                     Modbus_Address]: 1
[                    Inverter_Serial]: 2401081234
[                        Rated_Power]: 0
Traceback (most recent call last):
  File "/home/dividi/.local/bin/deye-read", line 8, in <module>
    sys.exit(read_from_inverter())
  File "/home/dividi/.local/lib/python3.9/site-packages/deye_controller/deye_reader.py", line 105, in read_from_inverter
    read_inverter(opts.address, opts.serial, batt_only=opts.battery, power_only=opts.power, combo=opts.combo,
  File "/home/dividi/.local/lib/python3.9/site-packages/deye_controller/deye_reader.py", line 46, in read_inverter
    string = '[{:>35s}]: {} {}'.format(reg.description.title(), reg.format(), suffix)
  File "/home/dividi/.local/lib/python3.9/site-packages/deye_controller/modbus/protocol.py", line 141, in format
    as_bytes = to_bytes(self.value)
  File "/home/dividi/.local/lib/python3.9/site-packages/deye_controller/modbus/utils.py", line 15, in to_bytes
    return bytes.fromhex(''.join([struct.pack('>h', x).hex() for x in val]))
  File "/home/dividi/.local/lib/python3.9/site-packages/deye_controller/modbus/utils.py", line 15, in <listcomp>
    return bytes.fromhex(''.join([struct.pack('>h', x).hex() for x in val]))
struct.error: 'h' format requires -32768 <= number <= 32767
[dividi@rocky deye-controller-master]$`

Is there any hint hot to get this working, possibly need to adapt some some to my inverter, but I don't yet have a ModBus manual that I'm sure is accurate for my inverter.

Thanks in advance.

Cheers.

githubDante, 2024-04-23:
Hi,

Looking at the traceback it fails on reading the system clock, which means that the register layout of this inverter is completely different. If you find the modbus documentation, please share it as I'm interested in expanding this lib.

Probably the sunsynk mapping for single phase inverters can be used. 
As you already have the lib installed, can you try to set the output from:

```console 
root@pi:~# deye-regtest 192.168.192.12 2913021234 22 3 
```

as a value to a DeviceTime instance and see if the formatted result:

```python

from deye_controller import HoldingRegisters
dt = HoldingRegisters.DeviceTime
dt.value = [6148, 6144, 3597]
dt.format()
```
matches the time shown by the inverter.

CidiRome, 2024-04-24:
Hello.

Reg 22 seems to have the Date and Time, it matches:

> [dividi@rocky ~]$ deye-regtest 192.168.192.12 2913028025 22 3
> [6148, 6154, 11018]

> 6148	24	4
> 6154	24	10
> 11018	43	10

Cheers.

CidiRome, 2024-04-24:
Hello.

I found this document elsewhere that seems to be accurate for my inverter, be aware that some registries return different values according what type of inverter you are connecting to. Also, I detected some descriptions are not very good and some translations that are there are also not accurate.

I made some tests with _deye-regwrite_ and _deye-regtest_ and I think I will manage to do what I need with it.

Cheers.
[sunsynk_modbus.docx](https://github.com/githubDante/deye-controller/files/15094196/sunsynk_modbus.docx)

githubDante, 2024-04-25:
Hi,

Basic support for single phase Hybrid/Micro/String inverters has been added in the [single-phase branch](https://github.com/githubDante/deye-controller/tree/single-phase). `deye-reader` has auto detection now. The registers list is still incomplete though.

CidiRome, 2024-04-25:
Hello.

Very nice, I confess that I'm learning how to read and write the registers from your code and not using it at all (seems I'm mostly using pysolarmanv5), but I'm happy to help in what I can.

Nevertheless, I thank you very much for the work you have done.

Changing the subject a bit, there are two registers I can't find in the document and that I need for my configuration:
[image]
[image]

By any chance, have you came across the register numbers for this two parameters?

Cheers.

githubDante, 2024-04-25:
Hi,

> By any chance, have you came across the register numbers for this two parameters?

See for changes in register 242.

CidiRome, 2024-04-25:
Hello.

Thank you for trying, that register changes this two values (at least the 2 most significant bits do):
[image]

By reading lots of 100 R/W registers at a time and changing the value and found the right one:
It is **326**: since the readings are 16 bit words, this two parameters are the most two significant bits of the less significant byte.
AC couple on Grid side is worth 64 and AC couple on Load side is worth 128.
So, to change this I would advise to read the current value change just those two bits and write it back, so not to change other bits that we don't know what are they for.

Cheers.

githubDante, 2024-04-26:
Hi,

If  I understand correctly register 242 controls the Grid/Gen signal options by changing the least significant bits:

```
0b0000000000000010 - 0x0001 - Grid signal enabled 
0b0000000000000001 - 0x0002 - Gen signal enabled 
0b0000000000000000 - 0x0000 - Both disabled
```

while for the _AC couple on_ options are controlled by register 326 by as follows:

```
0b0000000001000000 - 0x0040 - AC couple on grid side 
0b0000000010000000 - 0x0080 - AC couple on load side 
0b0000000000000000 - 0x0000 - Both AC options disabled
```
If the above is correct the AC couple options can be controlled from Python as:

Check (AND):

```python
ac_couple_on_grid_on = (inverter_value & 64) == 64
ac_couple_on_load_on = (inverter_value & 128) == 128
```

Toggle (XOR):

```python
ac_couple_on_grid_off = (inverter_value ^ 64)
ac_couple_on_load_off = (inverter_value ^ 128)
```

assuming that we work with the integers returned from `umodbus`.

CidiRome, 2024-04-26:
Hello.

Seems good to me.

I'm currently focusing on sending the data I read from the inverter to EmonCMS.

I already gathered all the values I want and put them in the order I want, I will now investigate how to construct the query to send them to EmonCMS.

Cheers.