githubDante/deye-controller #14: Generator Port Values and settings

markusdunckert, 2024-05-23:
Hi, 

I cannot access the GeneratorTotalPower in the Class HoldingRegisters. Could you please explain how to access the values here?
Thanks in advance

githubDante, 2024-05-23:
Hi,

Try with this

```python 
>>> from deye_controller import HoldingRegisters 
>>> from pysolarmanv5 import PySolarmanV5

>>> gen_total_p = HoldingRegisters.GeneratorTotalPower
>>> sol = PySolarmanV5('192.168.100.138', 27123456789, auto_reconnect=True)
>>> gen_total_p.value = sol.read_holding_registers(gen_total_p.address, gen_total_p.len)[0]
>>> gen_total_p.format()
0
```

githubDante, 2024-05-23:
With #15 it should be even easier 

```python
>>> from deye_controller import HoldingRegisters 
>>> from deye_controller.utils import monkey_patch
>>> from pysolarmanv5 import PySolarmanV5

>>> monkey_patch()
>>> gen_total_p = HoldingRegisters.GeneratorTotalPower
>>> sol = PySolarmanV5('192.168.100.138', 27123456789, auto_reconnect=True)
>>> sol.read_holding_registers(gen_total_p)
>>> gen_total_p.format()
0
```

markusdunckert, 2024-05-24:
Thank you for the advice, but unfortunately it doesnt work.
The first one breaks with:   gen_total_p = HoldingRegisters.GeneratorTotalPower
AttributeError: type object 'HoldingRegisters' has no attribute 'GeneratorTotalPower'

The second example: there is no function monkey_patch.

Any ideas?

githubDante, 2024-05-24:
Hi, 

Ensure that you are at the latest version 0.1.5 (`pip3 install --upgrade deye-conroller`). The grenerator related register were added after 0.1.4

As for the monkey_patch call it is not even merged in the main branch yet. It will be available in the next PyPI release.

markusdunckert, 2024-05-24:
great, that was the solution!

closed 2024-05-24 by markusdunckert