githubDante/deye-controller #17: Why max charge/discharge power is limited to 185A ?

toruonu, 2024-09-06:
""" Charge / Discharge """
    MaxChargeAmps = IntWritable(address=108, low_limit=0, high_limit=185)
    MaxDischargeAmps = IntWritable(address=109, low_limit=0, high_limit=185)

I'm curious why they are limited to 185? According to spec the SUN-12K-SG04LP3-EU (same as tested with, I have) has a max charge/discharge power of 240A. Should I just modify the code or is there a good reason for this?

Also as listed in another issue someone already found the sell mode register, maybe this upping to 240 and the new register could be added together to the next release or is it safe to just add myself :)

githubDante, 2024-09-07:
The limits are set according to the MODBUS documentation. It's available in the docs dir. The limits can be overwritten quite easily either through the _high attribute or by using a custom register. 

Register 142 will be included in the next version.

githubDante, 2024-10-11:
Support for register 142 has been added in 0.2.0

closed 2024-10-11 by githubDante