githubDante/deye-controller #18: TCP server implementation

nborisenko, 2024-10-10:
Hi, I try to connect wifi stick using server connect config_hide.html to my local tcp server, and I try to use this project as example, but no luck to decode, I see some correlation, but due 2 I get 2 types of messages(some data there in plain text, version, etc), other values I wasn't able to map. I ready want to use push model for avoid VPNs. Maybe already some spend some time on this, and have some inputs. If Author of this project don't have any observation I will put implementation to this project.

nborisenko, 2024-10-11:
Thank you.
I was try to use connection which logger establish to my server send to it payload like this (client  is PySolarmanV5):
           
           
            mb_request_frame = rtu.read_holding_registers(
                1, 185, 10
            )

            print("send mtu frame request")
            v5_request_frame = client._v5_frame_encoder(mb_request_frame)
            conn.sendall(mb_request_frame)
            
 But not any response get back looks like logger client, not process events like server.
 
 Yes, you are right PySolarmanV5 very nice library, add works good, I already try it.
 About proxy, this is definitely will works, but this is will required to put one more additional device in field, which will be problematically.

githubDante, 2024-10-11:
@nborisenko which option you change in config_hide.html 

1.  Server A Setting
2. Optional Server Setting
3.  Internal server parameters setting -> Protocol TCP-Client

githubDante, 2024-10-12:
It's 3. TCP-Client

See [here](https://github.com/githubDante/go-solarman-proxy)
Ready to use binaries can be downloaded from the GH Actions run [here](https://github.com/githubDante/go-solarman-proxy/actions/runs/11306459275) . See the Artifacts section at the bottom.

nborisenko, 2024-10-15:
You're just a magician. Thanks a lot. This is exactly what i need.

githubDante, 2024-10-15:
Glad to help.

closed 2024-10-15 by githubDante