githubDante/deye-controller #20: Modbus control of import and export power

HarmvandenBrink, 2024-11-05:
Would it be possible to control the import and export power of the DEYE inverter using Modbus?

Can we do this per phase, or as total amount?

Could you help me out with this, you’re sort of my last resort.

githubDante, 2024-11-05:
It's doable via the following registers:

 - MaxChargeAmps
 - MaxDischargeAmps
 or 
 - GridExportLimit 
 
 All are available in the WritableRegisters class. The latter is even mentioned at the bottom of the README.

HarmvandenBrink, 2024-11-05:
Sorry that I missed that. I thought it are limits that are set, not actual set points for the inverter

tonsmets, 2024-11-06:
I would like to know the same thing. With other systems I am used to writing a power setpoint (either a positive or negative value depending on discharge/charge).

githubDante, 2024-11-06:
> So I think my question is: if I just want to charge or discharge the battery right now with a certain amount of power, what commands do I need to send? This seems like a very basic functionality but I cannot find the correct commands/steps when looking at the documentation.

You should use the programmer with the correct time interval, power & SOC settings. Also make sure that the TimeOfUse (register 146) shows Enabled and at least the current day of the week is also in the list.

TimeOfUse example from `deye-read`:

```
[                   Sell_Time_Of_Use]: ['TimeOfUse - ENABLED', 'TimeOfUse - MONDAY', 'TimeOfUse - TUESDAY', 'TimeOfUse - WEDNESDAY', 'TimeOfUse - THURSDAY', 'TimeOfUse - FRIDAY', 'TimeOfUse - SATURDAY', 'TimeOfUse - SUNDAY'] 
```

davmoz, 2025-03-24:
Specifically regarding import, is setting the `MaxChargeAmps` the only way of doing this? 

What if I want to charge at exactly *3450W*? The resulting import power would be:

$P = \text{BatteryVoltage} \times \text{MaxChargeAmps}$

And this basically means that I will have to live with $\pm\frac{\text{BatteryVoltage}}{2}$ (ish) in worst case scenario, no?

Registers (holding): 
108 - MaxChargeAmps (R/W)
10003 - Battery Voltage (R)

githubDante, 2025-03-24:
Hi,

Which inverter is the cited documentation for ?
`MaxChargeAmps` looks like the only way. Registers 210 & 212 are not an option as they are controlled by the BMS (at least in my case). Any value written there is overwritten after a few seconds with no effect on the charge/discharge rate. 

You need 61-62 amps with a 56V  battery in order to charge it with 3450W/h (${P}/{BatteryVoltage}$).

davmoz, 2025-03-25:
> Which inverter is the cited documentation for ?

Apologies, here comes the setup and the doc: 

Inverter(s): SUN-5/6/8/10/12/15/20/25K-SG01HP3-EU-AM2
Battery module: GB-LM4.0
Docs (translated with DeepL): [deye_MODBUS RTU_V105.3-20240710-Lee_Dobson en.pdf](https://github.com/user-attachments/files/19443912/deye_MODBUS.RTU_V105.3-20240710-Lee_Dobson.en.pdf)

Also, I'm reading 6365.0 from register 587, so 636.50V.

githubDante, 2025-03-25:
So, this is a new version of the RTU documentation with a few additions for high voltage batteries/inverters. Thanks for sharing it. Considering the notes in the changelog (`Add customized register`) and the position of these registers (231-233) they are probably BMS related. 233 does not accept values in the LP/LV version too.