githubDante/deye-controller #22: Set (or sync) time on Inverter

ciciban, 2025-03-12:
### Pls. any description or examples how to set Date and Time on Inverter.

> register = WritableRegisters.DeviceTime
>register.set("2025-03-12 21:19:00")
>inv.write_multiple_holding_registers(register.address, [register.modbus_value])

and got error:

> register.set("2025-03-12 21:19:00") 
>  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> File "/pythonVE/lib/python3.11/site-packages/deye_controller/modbus/protocol.py", line 776, in set
>    as_ints = [int(u) for u in x.strftime('%y %m %d %H %M %S').split(' ')]
>                              ^^^^^^^^^^

I also extended class `DeviceTimeWriteable`  (in protocol.py File) and added `datetime.datetime.now()` , but no success.

>class DeviceTimeWriteable(WritableRegister):
>    """
>    Adjustments of the system time of the inverter
>    """
>    def __init__(self):
>        super(DeviceTimeWriteable, self).__init__(62, 3)
>    def set(self, x: datetime.datetime):
>        """ Set the time of the inverter """
>        **x=datetime.datetime.now()**
>        as_ints = [int(u) for u in x.strftime('%y %m %d %H %M %S').split(' ')]
>        self.modbus_value = to_inv_time(as_ints)
>        self.value = x

and got error:

>Traceback (most recent call last):
>  File "/home/solaru/pythonVE/lib/python3.11/site-packages/umodbus/functions.py", line 1515, in values
>    struct.pack(">" + conf.MULTI_BIT_VALUE_FORMAT_CHARACTER, value)
>struct.error: required argument is not an integer
>During handling of the above exception, another exception occurred:
>
>Traceback (most recent call last):
>  File "/xxxxxx/read-set-registar.py", line 33, in <module>
>    inv.write_multiple_holding_registers(register.address, [register.modbus_value])
>
>  File "/pythonVE/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 668, in write_multiple_holding_registers
>    mb_request_frame = rtu.write_multiple_registers(
>                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>  File "/pythonVE/lib/python3.11/site-packages/umodbus/client/serial/rtu.py", line 169, in write_multiple_registers
>    function.values = values
>    ^^^^^^^^^^^^^^^
>  File "/pythonVE/lib/python3.11/site-packages/umodbus/functions.py", line 1517, in values
>    raise IllegalDataValueError
>umodbus.exceptions.IllegalDataValueError:  The value contained in the request data field is not an allowable value

githubDante, 2025-03-12:
Hi,

In your first attempt pass a datetime object to the set method (i.e. `register.set(datetime.datetime.now())`) and then send it to the inverter with `inv.write_multiple_holding_registers(register.address, register.modbus_value)`

I should admit that the examples are ambiguous about this one.

ciciban, 2025-03-13:
Thx a lot...it's working

> import datetime
>
>register = WritableRegisters.DeviceTime
>register.set(datetime.datetime.now())  
>inv.write_multiple_holding_registers(register.address, register.modbus_value)

closed 2025-03-13 by ciciban