githubDante/deye-controller #24: fix for the commandline  deye-read or deye-regwrite no Socket Network connection error

maxxx999, 2025-05-16:
Hi,

i fixed  a missing errorhandling in the commandline  if there is no connection to defined adress 

python error - if missing networkconnection for address  example:

 ```
 File "/home/Aehtnix/Dokumente/deyecontroller/venv/bin/deye-read", line 8, in <module>
    sys.exit(read_from_inverter())
             ~~~~~~~~~~~~~~~~~~^^
  File "/home/Aehtnix/Dokumente/deyecontroller/venv/lib/python3.13/site-packages/deye_controller/deye_reader.py", line 105, in read_from_inverter
    read_inverter(opts.address, opts.serial, batt_only=opts.battery, power_only=opts.power, combo=opts.combo,
    ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                  as_json=opts.json, to_file=opts.out)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Aehtnix/Dokumente/deyecontroller/venv/lib/python3.13/site-packages/deye_controller/deye_reader.py", line 11, in read_inverter
    inv = PySolarmanV5(address, int(logger_serial), port=8899, mb_slave_id=1, verbose=False, socket_timeout=10,
                       error_correction=True)
  File "/home/Aehtnix/Dokumente/deyecontroller/venv/lib/python3.13/site-packages/pysolarmanv5/pysolarmanv5.py", line 133, in __init__
    self._socket_setup(kwargs.get("socket"), kwargs.get("auto_reconnect", False))
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Aehtnix/Dokumente/deyecontroller/venv/lib/python3.13/site-packages/pysolarmanv5/pysolarmanv5.py", line 487, in _socket_setup
    raise NoSocketAvailableError("No socket available")
pysolarmanv5.pysolarmanv5.NoSocketAvailableError: No socket available
Error: Fehler beim Ausführen von "deye-read --json 10.42.0.213 2100000000": Command failed: deye-read --json 10.42.0.213 2100000000
Traceback (most recent call last):
  File "/home/gehtnix/Dokumente/deyecontroller/venv/bin/deye-read", line 8, in <module>
    sys.exit(read_from_inverter())
             ~~~~~~~~~~~~~~~~~~^^
  File "/home/Aehtnix/Dokumente/deyecontroller/venv/lib/python3.13/site-packages/deye_controller/deye_reader.py", line 105, in read_from_inverter
    read_inverter(opts.address, opts.serial, batt_only=opts.battery, power_only=opts.power, combo=opts.combo,
    ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                  as_json=opts.json, to_file=opts.out)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Aehtnix/Dokumente/deyecontroller/venv/lib/python3.13/site-packages/deye_controller/deye_reader.py", line 11, in read_inverter
    inv = PySolarmanV5(address, int(logger_serial), port=8899, mb_slave_id=1, verbose=False, socket_timeout=10,
                       error_correction=True)
  File "/home/Aehtnix/Dokumente/deyecontroller/venv/lib/python3.13/site-packages/pysolarmanv5/pysolarmanv5.py", line 133, in __init__
    self._socket_setup(kwargs.get("socket"), kwargs.get("auto_reconnect", False))
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Aehtnix/Dokumente/deyecontroller/venv/lib/python3.13/site-packages/pysolarmanv5/pysolarmanv5.py", line 487, in _socket_setup
    raise NoSocketAvailableError("No socket available")
pysolarmanv5.pysolarmanv5.NoSocketAvailableError: No socket available
```

###  i fixed it for me with a change in the deye_reader.py:

i validate networkconnection before useing the pysolarmanv5 lib
----------------------------------------------------------------------------------

```
from pysolarmanv5 import PySolarmanV5, V5FrameError
from .modbus.protocol import HoldingRegisters, BatteryOnlyRegisters, TotalPowerOnly
from .logger_scan import solar_scan
from argparse import ArgumentParser
from .utils import group_registers, map_response
import json
import socket

def is_address_reachable(address: str, port: int, timeout: int = 5) -> bool:
    """Check if the given address and port are reachable."""
    try:
        with socket.create_connection((address, port), timeout):
            return True
    except (socket.timeout, socket.error):
        return False

def read_inverter(address: str, logger_serial: int, batt_only=False, power_only=False, combo=False,
                  as_json=False, to_file=None):
    # Prüfen, ob die Adresse erreichbar ist
    if not is_address_reachable(address, 8899):
        print(f"Error: Address {address} on port 8899 is not reachable.")
        return

    inv = PySolarmanV5(address, int(logger_serial), port=8899, mb_slave_id=1, verbose=False, socket_timeout=10,
```
  

maybe that helps others too 
wish you a good day all

githubDante, 2025-05-29:
Thank you for pointing that out.

githubDante, 2025-05-29:
It's included in version 0.2.2. Already available on pypi.org